efabless/caravel_mpw-one #11: Caravele doesn't provide internal picorv32 interrupts!

jean-m-cyr, 2020-11-26:
The Caravel SOC design has one external interrupt pin, but doesn't seem to provide any accessible internal interrupt ports for internal wishbone slave peripheral user projects that would need them. It would be a shame to have to drive an extra output pin just to loop it back to the external interrupt input pin.

RTimothyEdwards, 2020-11-27:
You're wanting to have the user project be able to raise an interrupt on the managment SoC through the wishbone bus?

ohAitch, 2020-11-30:
I believe that is the feature request seen on slack, correct.

jean-m-cyr, 2021-01-26:
Not over the bus. Wishbone doesn't support it. More likely one or two unused wires into the interrupt controller. Something akin to the user_clock2 signal user project port.

RTimothyEdwards, 2021-09-01:
This issue was corrected on the caravel version for MPW-two, so I am closing this issue now.

closed 2021-09-01 by RTimothyEdwards