efabless/caravel_mpw-one #22: Cannot harden design using option 2 and no macros

SweeperAA, 2020-12-04:
I'm attempting to harden my design using option 2 (https://github.com/efabless/caravel/tree/master/openlane#option-2). When I attempt `make user_project_wrapper` I get several errors and warnings and it finally errors out with:

```
[ERROR]: Floorplanning failed
[ERROR]: module decred_top not found in /project/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged.lef
[ERROR]: Check whether EXTRA_LEFS is set appropriately
```

I don't have any macros in my design so all of the code is listed under `VERILOG_FILES`, I don't have any blackbox files or LEF or GDS imports.

If I create a similar script as a separate design (like option 1), I can get through that type of flow. It seems that the script execution between the two options are different, at least in the beginning.

The script being used is https://github.com/SweeperAA/skywater130_decred_miner/blob/master/openlane/user_project_wrapper/config.tcl

I've also modified the interactive.tcl as the instructions indicated (https://github.com/SweeperAA/skywater130_decred_miner/blob/master/openlane/user_project_wrapper/interactive.tcl).

SweeperAA, 2020-12-04:
If I remove `user_project_wrapper/interactive.tcl`, the flow gets passed the floor planning step, but then errors out during Generating PDN:

```
[INFO] [PDNG-0016] Power Delivery Network Generator: Generating PDN
[INFO] [PDNG-0016]   config: /project/openlane/user_project_wrapper/pdn.tcl
-errorcode NONE -errorinfo Error:\ pdn.tcl,\ 2\ can't\ read\ \"::env(_VDD_NET_NAME)\":\ no\ such\ variable\n\ \ \ \ while\ executing\n\"error\ \"Error:\ \[file\ tail\ \$error_sdc_file\],\ \$error_sdc_line\ \$error\"\"\n\ \ \ \ invoked\ from\ within\n\"if\ \[catch\ \{open\ \$filename\ r\}\ stream\]\ \{\n\ \ \ \ sta_error\ \"cannot\ open\ '\$filename'.\"\n\ \ \}\ else\ \{\n\ \ \ \ #\ Save\ file\ and\ line\ in\ recursive\ call\ to\ source.\n\ \ \ \ i...\"\n\ \ \ \ (procedure\ \"source_\"\ line\ 5)\n\ \ \ \ invoked\ from\ within\n\"source_\ \$filename\ \$echo\ \$verbose\ \"\n\ \ \ \ invoked\ from\ within\n\"source\ \$PDN_cfg\"\n\ \ \ \ (procedure\ \"init\"\ line\ 36)\n\ \ \ \ invoked\ from\ within\n\"init\ \$config\"\n\ \ \ \ (procedure\ \"apply\"\ line\ 4)\n\ \ \ \ invoked\ from\ within\n\"apply\ \$config\"\n\ \ \ \ (procedure\ \"pdngen::apply_pdn\"\ line\ 14)\n\ \ \ \ invoked\ from\ within\n\"pdngen::apply_pdn\ \$config_file\ \$verbose\ \" -code 1 -level 0 -errorline 9
[CRIT] [PDNG-9999] Unexpected error: Error: pdn.tcl, 2 can't read "::env(_VDD_NET_NAME)": no such variable
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/new_pdn.tcl |& tee >&@stdout /project/openlane/user_project_wrapper/runs/user_project_wrapper/logs/floorplan/pdn.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /project/openlane/user_project_wrapper/runs/user_project_wrapper/error.log

    while executing
"try_catch openroad -exit $::env(SCRIPTS_DIR)/new_pdn.tcl  |& tee $::env(TERMINAL_OUTPUT) $::env(pdn_log_file_tag).log"
    (procedure "gen_pdn" line 8)
    invoked from within
"gen_pdn"
    (procedure "run_floorplan" line 41)
    invoked from within
"run_floorplan"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "/openLANE_flow/flow.tcl" line 168)
Makefile:25: recipe for target 'user_project_wrapper' failed
make: *** [user_project_wrapper] Error 1
```

agorararmard, 2020-12-09:
@SweeperAA: The instructions were a bit lacking in the past, but they were updated in the current master. Please, try the new instructions out and inform us if you're still having any issues, here or on slack.

agorararmard, 2020-12-23:
@SweeperAA: with the updated documentation and the removal of the interactive script, following option 2 has become much simpler and those who tried it out on mpw-one-b didn't face any issues with it. Thus, could we close this issue?

closed 2020-12-23 by SweeperAA