efabless/caravel_mpw-one #41: LVS/CVC issues with caravel modules

d-m-bailey, 2021-02-23:
This issue is an on going list of the device level LVS and CVC issues with caravel modules.
origin/develop  commit 384a7d517a82c49a0c4411c8132ce3f97eebfad5
The indentation of the list is intended to show the hierachy.
Unless otherwise indicated, CVC was run on the netlist extracted from GDS.

| Cell                                    | LVS | CVC | 
|-----------------------------------------|-----|-----|
| caravel                                 |     |     | 
| + chip_io                                |     |     |
| + gpio_control_block                     | OK? (vssd matched to vssd1)  | OK  | 
| + mgmt_core                              |     |     |
| !+ DFFRAM                                | NG (spice extracted from gds is unusable) | NG (spice in spi/lvs unusable)    |
| !- digital_pll                           | OK     | OK     |
| + mgmt_protect                           | -   | OK  |
| !+ mgmt_protect_hv                       | NG (verilog missing decap) | -   |
| !+ mprj2_logic_high                      | OK  | -   |
| !- mprj_logic_high                       | OK  | -   |
| + simple_por                             |NG (verilog and gds contain different cells) |     |
| + sky130_fd_sc_hvl__lsbufhv2lv_1_wrapped | NG (library spice incorrect)  | OK    |
| + storage                                | NG (missing sram gl verilog/spice) | NG (Hi-Z inputs) |
| !- sram_1rw1r_32_256_8_sky130            | - (missing gl verilog/spice)  | -   |
| + user_id_programming                    | OK  | OK    |
| + user_project_wrapper                   | OK  | OK  |
| !- user_proj_example                     | OK  | -   |

ax3ghazy, 2021-02-24:
@d-m-bailey: Thanks!

I have renamed the old device names to the new sky130 equivalents in the SRAM spice netlist we got from OpenRAM. Please
check if this helps: https://github.com/efabless/sky130_sram_macros/blob/sky130_name_mapping/sram_1rw1r_32_256_8_sky130/sram_1rw1r_32_256_8_sky130.lvs.converted.sp

d-m-bailey, 2021-03-05:
For running device level LVS and CVC on other projects on the MPW (assuming `user_project_wrapper` is the only difference).

1. cd _project_
2. Uncompress `gds/user_project_wrapper.gds`
2. Recursively check for referenced modules in `verilog/gl/user_project_wrapper.v`
`awk '/ \(/ {print $1}' verilog/gl/user_project_wrapper.v | sort -u | grep -v sky130_fd_sc_hd__`
3. For each submodule detected in 2, add an entry in `openlane/user_project_wrapper/config.tcl`
```
set ::env(LVS_EXTRA_GATE_LEVEL_VERILOG) "
  $script_dir/../../verilog/gl/submodule1.v
  $script_dir/../../verilog/gl/submodule2.v
  ..."
```
4. From docker, in _project_/openlane extract device level spice from layout.
```
flow.tcl -design user_project_wrapper -tag lvs -lvs -net ../verilog/gl/user_project_wrapper.v -gds ../gds/user_project_wrapper.gds
```
5. Create spi2cdl script
```
cat > spi2cdl <<'!'
# /bin/bash
  
mkdir -p $1/runs/lvs/results/cvc

awk '
/^X.* .*pr__[^ ]*fet.*=/ {
        print "M" $0; 
        next;   
}
/^X.*pr__diode/ {
        print "D" $0; 
        next;   
}
/^X.*pr__cap_mim_m3_. / { 
        print "C" $0; 
        next;   
}
/^X.*pr__res_xhigh_po / {
        $4 = "";
        print "R" $0; 
        next;   
}
/^X.*pr__res_generic_.d/ {
        $6 = sprintf("$SUB=%s %s", $4, $6);
        $4 = "";
        print "R" $0; 
        next;   
}
 { print $0; }
' $1/runs/lvs/results/magic/$1.gds.spice > $1/runs/lvs/results/cvc/$1.cdl
!
```