efabless/caravel_mpw-one #64: make run-precheck hangs at XOR check

techeraj-osh, 2021-06-15:
The content of checks/xor.log while running "make run-precheck" 

First Layout: /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper_empty_erased.gds
Second Layout: /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper_erased.gds
Design Name: user_project_wrapper
Output GDS will be: /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper.xor.gds
Reading /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper_empty_erased.gds ..
**ERROR: In /usr/local/bin/xor_checks/xor.drc: Stream has unknown format:** /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper_empty_erased.gds in Layout::read
**ERROR: Stream has unknown format:** /home/s2s-svr1/projects/asic/trial/caravel_user_project/checks/user_project_wrapper_empty_erased.gds in Layout::read in MacroInterpreter::execute
  /usr/local/bin/xor_checks/xor.drc:15:in `execute_drc'
  :/built-in-macros/drc_interpreters.lym:18:in `instance_eval'
  :/built-in-macros/drc_interpreters.lym:18:in `execute_drc'
  :/built-in-macros/drc_interpreters.lym:92:in `execute'

techeraj-osh, 2021-06-15:
Any suggestion to solve this issue  would be really help full

RTimothyEdwards, 2021-09-01:
If I recall, this was a klayout issue and was fixed.  I'm going to close this issue on the belief that it has been resolved.  If that is not the case, please feel free to reopen it.

closed 2021-09-01 by RTimothyEdwards