efabless/caravel_mpw-one #66: Gate Sim Issue: user_project_wrapper Port name mismatch between RTL & Gate level caravel.v

dineshannayya, 2021-07-05:
We see user_project_wrapper Port name mismatch between RTL & Gate level caravel.v 

In RTL : caravel/verilog/rtl/caravel.v 
	user_project_wrapper mprj ( 
                 ....
		.**la_oenb**(la_oenb_user),

in Gate Level: caravel/verilog/gl/caravel.v 
     user_project_wrapper mprj (
              ......
            .**la_oen**(

In user_project_wrapper has **la_oenb** port, where as gate level caravel.v has **la_oen**

dineshannayya, 2021-07-05:
Sorry wrong close this only .. This is still a issue.

RTimothyEdwards, 2021-09-01:
This remained an issue until the final synthesized layouts for Caravel were completed for MPW-two.  Those were finished a bit late but should be good now.  If you believe that that is not the case, please feel free to reopen the issue.

closed 2021-09-01 by RTimothyEdwards