efabless/caravel_mpw-one #69: Synthesis error with openlane

ankitwahane, 2021-08-04:
1. Executing Verilog-2005 frontend: /openLANE_flow/designs/serv/src/test2.v
Parsing SystemVerilog input from `/openLANE_flow/designs/serv/src/test2.v' to AST representation.
Generating RTLIL representation for module `\$paramod\serv_ctrl\RESET_STRATEGY=1296649801\RESET_PC=0\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_mem_if\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_rf_if\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_state\RESET_STRATEGY=1296649801\WITH_CSR=1'.
Generating RTLIL representation for module `\serv_alu'.
Generating RTLIL representation for module `\serv_bufreg'.
Generating RTLIL representation for module `\serv_csr'.
Generating RTLIL representation for module `\serv_decode'.
Generating RTLIL representation for module `\serv_immdec'.
Generating RTLIL representation for module `\serv_top'.
Successfully finished Verilog frontend.

2. Executing Verilog-2005 frontend: /openLANE_flow/designs/serv/src/serv_bufreg.v
/openLANE_flow/designs/serv/src/serv_bufreg.v:1: ERROR: Re-definition of module `\serv_bufreg'!
[ERROR]: during executing: "yosys -c /openLANE_flow/scripts/synth.tcl -l /openLANE_flow/designs/serv/runs/test_ex/logs/synthesis/yosys.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check yosys  log file
[ERROR]: Dumping to /openLANE_flow/designs/serv/runs/test_ex/error.log

ankitwahane, 2021-08-12:
please help me with this...i am stuck on this step from so long

Piyush3dB, 2021-08-12:
What does the log file say?

ankitwahane, 2021-08-12:
> What does the log file say?

this is what is written in the log file

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3621 (git sha1 84e9fa7, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.

1. Executing Verilog-2005 frontend: /openLANE_flow/designs/serv/src/test2.v
Parsing Verilog input from `/openLANE_flow/designs/serv/src/test2.v' to AST representation.
Generating RTLIL representation for module `\$paramod\serv_ctrl\RESET_STRATEGY=1296649801\RESET_PC=0\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_mem_if\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_rf_if\WITH_CSR=1'.
Generating RTLIL representation for module `\$paramod\serv_state\RESET_STRATEGY=1296649801\WITH_CSR=1'.
Generating RTLIL representation for module `\serv_alu'.
Generating RTLIL representation for module `\serv_bufreg'.
Generating RTLIL representation for module `\serv_csr'.
Generating RTLIL representation for module `\serv_decode'.
Generating RTLIL representation for module `\serv_immdec'.
Generating RTLIL representation for module `\serv_top'.
Successfully finished Verilog frontend.

2. Executing Verilog-2005 frontend: /openLANE_flow/designs/serv/src/serv_bufreg.v
Parsing Verilog input from `/openLANE_flow/designs/serv/src/serv_bufreg.v' to AST representation.
/openLANE_flow/designs/serv/src/serv_bufreg.v:1: ERROR: Re-definition of module `\serv_bufreg'!

Piyush3dB, 2021-08-12:
> 2\. ERROR: Re-definition of module

So this is the problem with your code right?...
`
/openLANE_flow/designs/serv/src/serv_bufreg.v:1: ERROR: Re-definition of module 
`

ankitwahane, 2021-08-12:
> > 2. ERROR: Re-definition of module
> 
> So this is the problem with your code right?...
> `/openLANE_flow/designs/serv/src/serv_bufreg.v:1: ERROR: Re-definition of module `

but the code is so simple. the module is defined only once.

module serv_bufreg
  (
   input wire 	      i_clk,
   //State
   input wire 	      i_cnt0,
   input wire 	      i_cnt1,
   input wire 	      i_en,
   input wire 	      i_init,
   output reg [1:0]   o_lsb,
   //Control
   input wire 	      i_rs1_en,
   input wire 	      i_imm_en,
   input wire 	      i_clr_lsb,
   input wire 	      i_sh_signed, 
   //Data
   input wire 	      i_rs1,
   input wire 	      i_imm,
   output wire 	      o_q,
   //External
   output wire [31:0] o_dbus_adr);

   wire 	      c, q;
   reg 		      c_r;
   reg [31:2] 	      data;

   wire 	      clr_lsb = i_cnt0 & i_clr_lsb;

   assign {c,q} = {1'b0,(i_rs1 & i_rs1_en)} + {1'b0,(i_imm & i_imm_en & !clr_lsb)} + c_r;

   always @(posedge i_clk) begin
      //Make sure carry is cleared before loading new data
      c_r <= c & i_en;

      if (i_en)
	data <= {i_init ? q : (data[31] & i_sh_signed), data[31:3]};

      if (i_init ? (i_cnt0 | i_cnt1) : i_en)
	o_lsb <= {i_init ? q : data[2],o_lsb[1]};

   end

   assign o_q = o_lsb[0] & i_en;
   assign o_dbus_adr = {data, 2'b00};

endmodule

ankitwahane, 2021-08-13:
> > 2. ERROR: Re-definition of module
> 
> So this is the problem with your code right?...
> `/openLANE_flow/designs/serv/src/serv_bufreg.v:1: ERROR: Re-definition of module `

done checking. thank you piyush for your help. problem was another .v file was using this module in the same directory. removing it solved the problem.

closed 2021-08-13 by ankitwahane